frntc/SIDKick-pico #5: Pico W for extras?

mwedmark, 2024-01-11:
Does the implementation allow for some extras with WIFI and/or Bluetooth functionality in the future?
Any reason for a PiPico W not to work?

For example, some ideas:
- Sound output through BT directly to speaker?
- Web-UI for changing settings live while playing?
- Output to external WIFI board (f.e. ESP8266) to control LED's gor graphics feedback?
- Mouse control via either websocket or via BT-mouse?

magwed, 2024-01-12:
Also this would make it possible to send the input/control values to SID to external listener for debugging/playing on f.e. a PC for reference! :-)

frntc, 2024-01-12:
I have thought about streaming audio via BT, WLAN or USB (as a USB-microphone), but not about the other options. I might look into that at some point, but I don't know yet whether any/which of these functionalities impact the running bus communication and how much resources they consume (and when/if I might be trying things out, streaming audio would be higher on the list than others).

About the mouse: supporting USB mice is something I have in mind since years (already for the large SIDKick), but the "annoying" part is that mouse buttons require connections to the joystick port/CIA, i.e. it requires additional connections inside the C64 (and thus GPIOs).