frntc/SIDKick-pico #10: SKP sound strange on C64 NTSC

mwedmark, 2024-01-18:
I've just build my first good Sidkick pico and using latest firmware 1.14. I tried it first on my odd NTSC machine just to be sure. All music sound weird and ill tuned. It still "plays" the tune but something with the timing/bus must be wrong. Also the settings menu was not possible to visit either so no settings could be made. Should the NTSC be auto-checked or is it up to the user to change it? When moving the exact same SKP to my childhood machine (been mine for 35 years) it sound perfect! So cool, thanks for sharing!

Is there anything I can do to help out? Coding for now I'm not ready yet but testing specific programs, recording video/audio, PEEK/POKE, testing versions of firmware

frntc, 2024-01-18:
I've tested the SKpico on NTSC machines, but it might still be possible that the timing for some machines should be different (but your's would be the first one). Out of curiosity you could check if paddles/mouse is working (this would tell if reading values from the SKpico works).

The SKpico *config tool* detects PAL vs NTSC, i.e. you need to run it once. 

Patching the timing value is something that's prepared: search for 01 02 03 04 05 06 (in hex editor) in the .uf2 -- the two bytes before that are the delays after phi2 and thereafter until a value is put on the bus.

mwedmark, 2024-01-21:
I gonna try and look into this today.
Starting reference:
To measure the PHI2 clock I've started by using the Diag-tool found in Kung Fu Flash as a guide.
My PAL machine measures after 56 measurement average: 985'218Hz while official says 985'248 Hz so about 0,003% slower.
My NTSC machine measures after 600 meas. average: 1'022'693Hz while official says 1'022'727Hz so also about 0,003% slower.
Both of them aquire this value without flutuation within a few measurements. I'm not sure how exact that number is really.
Both computers are stable with Kung Fu Flash as some kind of reference.
Next test will be trying out the mouse in the menu of the NTSC computer.
Same mouse has been tested in my PAL machine and gives values but does produce a lot of jitter which currently makes the mouse unusable, but you can see the mouse move in the correct direction.

mwedmark, 2024-01-21:
https://github.com/frntc/SIDKick-pico/assets/2123085/f22a6d26-35fe-4dbd-ba85-af2d5ed94a17

Here's a recording from a unchanged SKP in my C64 NTSC. Some stuff sound much worse than this so this might be a good starting point. It is not possible to start menu, it just crashes currently. I'm gonna play around with the timing mentioned above. original settings for firmware 0.14 seem to be: 0B, 0F so I guess I'll make small adjustments and see what happens.

mwedmark, 2024-01-21:
Problem solved! Sorry to say, my socket on the NTSC machine is not making perfect contact with all pins so some specific data bit is not coming through. I've tried 8 different timing schemes in which several did make the sound go away completely and 100% silent. In the other settings it sound exactly the same so it does not seem to be really nitpicky on timing. I'm using rounded pins which may actually be smaller/thinner than original so I need to fix that in the socket. Otherwise all good. I'm closing this bug shortly. NFF in firmware.

closed 2024-01-22 by frntc