frntc/SIDKick-pico #11: STILL... A serious problem

Braincell1973, 2024-01-20:
MUTANTS & SHADOWFIRE are STILL not right at all, not even close.

MUTANTS seems to repeat chunks of melody........

Shadowfire still has the seem issue i reported in 0.12 & 0.13.

Also, There seems to be some strange sound artifacting on some tunes where its like you can here the channel close with a dull POP....

0.14, using oneload64v4 via KFF cart

PWM

frntc, 2024-01-20:
I would not have closed the issue if it would not have worked here (I compared to Vice and it sounded the same), maybe there is something else going on with your hardware. I'm happy to check out some recordings if you can send them (best via mail).

Braincell1973, 2024-01-20:
Ah , i see, OK, I will try it on a different machine and use something other than a KFF

Braincell1973, 2024-01-21:
OK, so tried it on a C64c with the samer KFF and files and it worked perfectly...... my breadbox must have partially shat itself, strange as many tunes play properly.

The channel closing "DOOF" still exists though and that is not in the real hardware.

To recreate - load enigma force from the oneload64 collection, at the beginning when the writing is going accross the top of the screen, at the end of each line you can hear a low "doof" type sound like the channel is closing.

Does not do this on real 6581 or 8580

cheers lee

frntc, 2024-01-22:
I have compared the SKpico-output of Enigma Force to that of reSID in Vice. Both are identical, but yes, there are pop-sounds after a line is printed. Again, if you have recordings, I'm happy to check them out.

Braincell1973, 2024-01-22:
I am making a youtube video, will try to augment, its not serious but its sometimes noticeable.

Will send you a link when I get it all edited

Braincell1973, 2024-02-16:
https://youtu.be/71Z10nYyaZ8?si=_ws1Hupz59SrRxQk

frntc, 2024-02-17:
Hey, can you point me to the time stamps where I should listen closely?

Braincell1973, 2024-02-17:
Start watching from 12mins

frntc, 2024-02-20:
You mean the sounds you flagged with "There!!!"? This seems to be due to the wave forms as they are implemented in reSID 0.16 (the SID emulation used in SKpico). Maybe I'll release a firmware with an improved reSID tailored for SKpico which I have here (don't get me wrong, reSID 1.0 exists since years and is much more accurate, but it's too compute/memory intensive for the Pico).

If there's no other "bug", I'd close this issue!?

Braincell1973, 2024-02-20:
Hi
     No other bug, but, as I demonstrated in the video the breadbin with a real sid didn't have the problems the pico did, still could be the breadbin as it's a piece of shit.....

Yes the "there" sound is what I was talking about, its a really minor point as the pico simply cannot be beaten for cost ......

Please close the issue, cheers for your efforts.

Lee

frntc, 2024-02-21:
The most common problem is a bad socket for the IC/SID and thus unreliable contact. I recommend checking that in your breadbin.

closed 2024-02-21 by frntc