frntc/SIDKick-pico #12: Problems calibrating analog mouse

mwedmark, 2024-01-21:
I'm using SKP with latest firmware together with an analog mouse that I know works with an old 6581 in f.e. Arkanoid.

The mouse is a M-3 Mouse (Contriever brand). I've tried most of the different settings but the "pico" sprite jumps around a lot. With the setting of "smooth mouse", disabled "level" and disabled "outlier", it seem to be pretty stable at the middle of the screen when not moving the mouse, I see values of 7A-80 in X and values around 6D-7x for Y. When I move it, the values/sprite jumps around a lot, even with this settings. In Arkanoid you can, kind of, control it (think it is going backwards..) but because of the jumping it is unplayable.

I can try more stuff but unclear what to fiddle with.

I also think the mouse can be used in different modes (can work as Joystick, Analog or Paddle mode) but this mode works perfectly with the original SID.

frntc, 2024-01-22:
I don't have one of these mice, but I think one of the testers had one. Some mice have a quite high "probe"-signal on the potentiometer line to sync with the SID, and this might be misinterpreted as the rising edge of the mouse-signal. To compensate for this, you should enable "level", and try "outlier" (there are two intensities). If that doesn't help, I'd be curious to look at an oscilloscope measurement ;-) I've also heard of people who modified the mouse-hardware to lower this signal. At least for 1351s it seems that each device has an (almost) individually chosen set of resistors to adjust everything...

magwed, 2024-01-22:
I tried all different settings available in the menu and by the look of how the Pico sprite moves, nothing helps. The computer is the original one from my childhood and using the original SID. The mouse was won at an online auction just half a year ago. Mouse seem to be in perfect shape including original box, original manual and disk. Mouse has hardly been used, I would say. When using the original stuff together with the mouse it works flawless in Arkanoid, paint programs and GEOS. I will also try it on my NTSC-machine with pico so see if anything changes.

After that, I think we should move to next level with this one. I have a Bitscope BS-50 shat I could use, no problem. I would just need to know where/what to measure. Do you want me to measure on the PotX/PotY pins directly on the SKP? Or maybe I need to introduce controllable resistor on the outside to be able to measure at different Ohm's?

mwedmark, 2024-01-22:
I've setup my Bitscope. Is there a Schematics to be able to track back exactly which pin on Pico which is connected to each pin on the SID? Would be nice with a simple Schematics when doing fiddling with it. I've looked in the code and found out some pins:
#define D0			0
#define A0			16
#define A5			14
#define A8			15
#define OE_DATA		8
#define RW			9
#define PHI			12
#define AUDIO_PIN	13
#define SID			21
#define LED_BUILTIN 25

frntc, 2024-01-22:
Please try to measure the signal before the SKpico first...

mwedmark, 2024-01-23:
Test done using Final Cartridge III Desktop where mouse movements are perfect while measuring with original circuit and moving very irregular with SKP. If needed I can supply short movies of the different behaviours, but let's start here.

Measurement done on pot-Y (pin 23 on original SID circuit 6581 from 1983). 2 pictures is added here. One shows the minimum length of high tip produced by my M3-mouse before it trips over to max. The other picture shows the maximum length of high tip before it trips over to min.

Some text that I tried to interpret the values between takes written below. Take them all with a grain of salt.

Low level: 0.1V-0.2V
High level: 2.5V? (Should be 4-5V so maybe scope is doing something funky, bad ground?)
peak-2-peak: 2.3V-2.4V
Full period length: 200us/div => 2.4 div period 480us/period

Time at/above low value (sum of low porch + high porch):
1.2 div => 240us high and 240 us low

When getting the highest value from mouse:
Length low porch: 0.5 div => 100us
Length high porch: 0.7 div => 140us
80us+160us = 240us above low value

When getting the lowest value from mouse:
Length low porch: 1 div => 200us
Length high porch: 0.2 div => 40us
200us+40us = 240us above low value

The only thing different here is the length of the high porch which differ between from 140us to 40us in length. This different will be the full span of the mouse function for pot-Y. Is this enough to be able to check something?

[image]
[image]

frntc, 2024-01-23:
The timings look good, standard behavior. The voltages are more interesting. Some mice have this plateau before the actual signal (the high signal) -- which could wrongly be detected is "high". Estimating from the images (= assuming the high signal is 5V) the plateau is about 1.5V to 1.6V -- the "level" option in the config should normally be able to pull this down (at least it did for mice with a similar behavior). Maybe you can try to measure the actual voltage levels.

mwedmark, 2024-01-23:
Level did not help, neither did outlier. Mouse pointer jumping all around still, sorry to say.

Same measurements but with cursor enabled on my Bitscope. Period seem to be very close to 500us which is good.
p-t-p low porch seem to be very close to 800mV with my current scaling. So by scaling 2.5V=>5V this would mean 1.6V, just like you said.

[image]

frntc, 2024-01-24:
Maybe that 1.6V is on the edge. There are some experiments you could do to tame this mouse. But as it (luckily/hopefully) is a rare phenomenon (and because the experiments might be dead ends or just for the sake of completeness) I'd like to take this offline. Want to write me a mail?

mwedmark, 2024-01-24:
PM'd you at Lemon64. My Googlefu was low today so I didn't find the email address.. :-)

frntc, 2024-01-24:
:+1: 

I'll close this issue for now, and will re-open once/if there are findings to report.

closed 2024-01-24 by frntc

r1me, 2024-03-12:
I use [M]ouse by svofski (A PS/2 to 1351 adapter) and can confirm that mouse cursor jumps all over the place which does not happen with the original SID.

mwedmark, 2024-03-12:
We did some good research but did not fix this issue. My mouse is still not working with SKP but perfect on 2 different machines. I may have one guess (that I found today actually..), since I think that this has been tested successfully on other machines.
Look at this link: https://www.facebook.com/groups/commodoresixtyfour/permalink/10159433568132172

Similar problems with paddles (same thing as analog mouse I guess). The problems I see is a very old original 5-pin Video Breadbox version which could potentially be the same thing, bad/non-matched capacitors on motherboard. What model are you using when you get this behaviour? Still they work with my original SID 6581.. but maybe outside of general spec?

frntc, 2024-03-12:
I'm also using [M]ouse without any problems, not yet sure what the interplay of mice/adapter/board is that causes trouble. I have some ideas and will do some more experiments. We'll see what comes out.

mwedmark, 2024-03-23:
@r1me could you please try and run Westbank and listen to the music there? For some unknown reason, this particular game causes problems when using the SID. Notes keep dropping both through menu and in game. It would be interesting to see if you also have the same problems here.

mwedmark, 2024-05-18:
Just to follow up, latest firmware 0.20 tested right now. With O/V enabled (outlier and level enabled) the mouse moves super smooth. Really great work with the updates. This is great stuff!