frntc/SIDKick-pico #22: reSID+digi detect actually mixed up with reSID 0.16 ?

r1me, 2024-03-12:
I configured my SKpico for 2x6581 and was switching digi samples "heuristic" on/off and noticed huge sound difference in a third track of Vicious SID demo. However with the option enabled (reSID+digi detect) it sounds worse (my opinion) with high frequencies cut out, almost like enabled got mixed up with disabled in config tool UI. Just to make sure I ran Vicious SID on my 6581 equipped breadbin, and it sounds practically the same as SKpico but with emulation engine set to reSID 0.16, not +digi detect.

Recorded few sec:
[vicious digidetect.zip](https://github.com/frntc/SIDKick-pico/files/14575527/vicious.digidetect.zip)

frntc, 2024-03-12:
The "digi detect"-option is based on heuristics which might not cover every case correctly, but I think it does in principle here. It then bypasses the waveform generation and directly outputs the sample value to the subsequent filters. If it sounds ok without the heuristics, I'd say you don't need it :-)

r1me, 2024-03-12:
Ok, my bad. I thought "on" was mixed with "off" somewhere in code or config tool GUI. Simply because I expected reSID+digi detect sound closer to original SID, but in my opinion it does opposite, at least with mentioned Vicious SID demo ("Hunters Moon" track to be specific). I will recommend to keep it off in article.

closed 2024-03-12 by r1me

frntc, 2024-03-13:
These heuristics are meant as an alternative path to reSID 0.16 (which does not support all new techniques) and is meant to be replaced once I improved the emulation (e.g. by integrating/adapting things from reSID 1.0)