frntc/SIDKick-pico #23: Sound on westbank sounds very strange on SKP

mwedmark, 2024-03-14:
For me,  trying out several games ask around good but Westbank sound very choppy. It doesn't matter if I choose 6581, 8580 or 8580+dig. They ask around like crap. Can someone else try it too? I also tried 2 different versions of the game, both sound bad.

mwedmark, 2024-03-17:
Currently only thing connected to my machine is:
Kung Fu Flash v.1.49 which is loading the files
SKP latest firmware. No extra settings or cables enabled. Tones skipping both in PWM output to TV and DAC output.
Tried 2 different releases of the game, both sounds equally bad on this machine but those files sounds correct in VICE.
So that rules out bad release theory. All other games I've tried this far sounds really good. Not sure why this is different..

And the machine type is an old Breadbox:
C64 ASSY 250407 Rev A. 5-pin Video connector.

Just switched to the original 6581 SID, sound 100% perfect sound. I tried 2 different SKP both sounds bad.

frntc, 2024-03-21:
The game is launched as PRG, correct? Still, a recording would help...

mwedmark, 2024-03-21:
Thanks for reminding me. I tried one CRT-release and one PRG-release. Same problem with both. I have a short movie clip with same computer running both SKP and original chip for reference, I'll add them when I can get them to fit. I think I will do some crosstesting on some 10 random games to see if I hear something similar in other games.

Here are the links for Video showing/playing both:
SIDkick-pico:
https://www.youtube.com/watch?v=X670aniwcr0&ab_channel=MagnusWedmark
SID 6581:
https://www.youtube.com/watch?v=dQPylMuSeY0&ab_channel=MagnusWedmark

frntc, 2024-03-21:
Since all the SID-commands seem to be arriving and handled properly, this rather sounds like a problem of something analog (after the SKpico output) to me. How do you listen to the DAC output? Can you test with headphones?

mwedmark, 2024-03-21:
Thanks for spending time on this. Yes, same problem in DAC-output actually. I also thought about that but sounds just as bad.. I usually use DAC/headphones as I my C64 is close to the family TV. I would like to bring out my NTSC machine to double-check this in another machine too. Already tried 2 different SKP's as mentioned above.

frntc, 2024-03-21:
(just to be sure: I assume the SKpico is configured for NTSC in the config tool)

Do you directly connect the headphones to the DAC output?

mwedmark, 2024-03-21:
Yes it autochanges in the menu between NTSC/PAL when trying different machines. Yes I connect headphones directly to DAC. Both NTSC and PAL have the same problem for me. I also tried 2 different releases of the game and also 2 different power supplies. Always same problem. Tried several other games and they all sound great! Maybe we can't get any more clarity here until there are more machine getting the same problems. I don't get it!

frntc, 2024-03-22:
When the config tool is starting correctly, it's basically 100% guaranteed that the bus communication works (in the video I can also hear all the notes being played). This is why I suspect something after the SKpico not "harmonizing". Can you try setting a lower volume in the config tool?

mwedmark, 2024-03-23:
Thanks for the ideas! Nope Volume half level, still skipping some notes during the shole game. I tried even some more games like: 1942 och Uridium that I know how they should sound and they sounds perfect to me... Could this adhere from the same matter as the Jumping Mouse problem that I and another bloke had? Some kind of Voltage problem with mother board? The address lines are driven by buffer aren't they? Why one game?

frntc, 2024-03-24:
A voltage problem on your board would probably also affect other games, but making sure the power supply is good never hurts. The level shifters are pretty agnostic to some deviations here. Your mouse problem is not related to that (I suspect it's related to the 4066 and/or capacitors), still looking to get a non-working combination in my hands to check. 

To me it sounds that all notes are there, but the output at the end of the pipeline is not correct. Can you record audio directly from the line out of the DAC?

modeler, 2024-04-09:
I tested this with my set-up and have sound issues in West Bank also. With v0.14, channel 3 keep dropping out:

https://youtu.be/vQhQS-f5g00

With the v0.12 firmware, there are no voices, only pops. This is one of Fred Gray's tunes, tried a few other games that use his player and this is the only one I found that has issues (so far):

https://deepsid.chordian.net/?file=/MUSICIANS/G/Gray_Fred/

frntc, 2024-04-09:
Yes, you shouldn't use 0.12 unless you have strange paddle behavior as you (@modeler) do :)

I think they're not all the same problem: the "plop" (iirc Shadow Fire) comes from reSID 0.16 waveforms (in my working-version this is fixed), the other bug has overlaps (see below). Also it seems that the choppy sound (@mwedmark ?) is not the kind of artifacts that @modeler meant...

... I noticed that the behavior in Westbank is different between 6581 and 8580 (maybe you can verify, 6581 is fine here). What this player does is something like this: 
8a0c  LDA $8A9D,X
8a0f   STA $D404,X
8a12  INC $D404,X

Of course $D404 is a write-only register and thus this code relies on the value "still floating on the bus". This behavior is emulated is SKpico (with different fall off times for 6581 and 8580), but maybe I should relax this timing and/or make it configurable.

modeler, 2024-04-20:
I don't know how relevant this is, but with the current v0.14 firmware (from January) Fred Gray's tunes sound much better in my short board C64C than my long board breadbin. The issue #31 I initially raised only affects the breadbin, I wasn't able to pin down the actual cause yet.

It seems v0.14 is much more stable when using the short board, almost perfect in fact. I'll keep digging to see if I can find a long board that behaves and try to work out the differences.

frntc, 2024-05-17:
We had some tests and it should now be stable with the new firmware. For this reason I'll close the issue here (in case problems persist, please read the troubleshooting section in the readme, and if that doesn't help, open a new issue)

closed 2024-05-17 by frntc