frntc/SIDKick-pico #30: Clarification on license please.

dabonetn, 2024-04-01:
What's you position on reselling spare pcbs for cost + shipping?

I ordered extra bare boards from PCBWay, and was wondering if I could offer them up for the 0.66 each + the cost of a stamp and envelope?

frntc, 2024-04-01:
Thanks for asking -- dissemination of the project in this way is greatly appreciated! The license only forbids commercial intents (e.g. selling not just spare units on ebay, high volumes/prices, offer from persons who run shops).

closed 2024-04-01 by frntc

youngd24, 2024-08-28:
Another question related to this, I ordered 10 boards and used a couple myself, was thinking of making up a few kits and a few completed ones to sell at VCFMW, is that ok within the license? Kits for like $10-$12, assembled for $20-$25.