frntc/SIDKick-pico #41: DAC sound on some tracks has distortion...

Braincell1973, 2024-04-14:
Hi,
    Using the dual PWM & DAC firmware 0.14 i have noticed some tracks have distortion......

But on PWM output on the same dual firmware the sound is superb.......

https://youtu.be/HWDtWiZlBCA?si=cKPHV06s_XPNz3uW

This should help diagnose....

Lee

frntc, 2024-04-14:
This sounds as if Phi2-clock and that what the SKpico expects differ (DAC scratchy, PWM fine). I assume PAL/NTSC is set correctly in the config tool? Can you measure Phi2 in your machine?

modeler, 2024-04-20:
I tried the tune in question (https://hvsc.csdb.dk/MUSICIANS/M/Mitch_and_Dane/Dane/Seeress_Foretold_My_Future.sid) through the DAC audio on my short board C64C (0.14 DAC only, SKpico_DAC_LED.uf2) and it sounds perfect, no distortion at all.

Braincell1973, 2024-04-21:
I'm using the combined DAC + PWM on a c128.

Pwm is perfect, DAC is distorted...... weird

Other tracks great on DAC ???

Weird distortion is evident on my video I linked but I'm away with work this week so can't do anything till I return.

I will try another SK pico in one of my c64s, good to know the dac only works ok, DAN , are you able to try the pwm and dac firmware on your board?

Lee

modeler, 2024-04-22:
> I will try another SK pico in one of my c64s, good to know the dac only works ok, DAN , are you able to try the pwm and dac firmware on your board?

I tried it with the SKpico_v014/SKpico_DAC_PWM.uf2 firmware and it plays fine:

https://youtu.be/bZhRAtnbOvU

I do have a C128 myself though, will dig it out and see if I mine behaves the same.

modeler, 2024-05-01:
@Braincell1973 I was able to test the DAC firmware with my C128 today and the tune played fine (as a PSID executable, attached). Sorry for the delay, the darn thing had a bad RAM chip so had to repair that first!

I wonder if it is the demo itself that is causing the issue? I have seen one case where the audio in a demo is distorted with SKP (PWM and DAC), yet it plays as a stand-alone PRG quite happily. Might be worth you trying this tune on its own on your machine?

https://drive.google.com/file/d/1FNEtHFiG0NaTitvadJIvSaTmRiaBG085/view?usp=sharing

modeler, 2024-05-10:
Listening to a long SID tune via the DAC output, I somehow managed to hear the same sort of distortion. It crept in after about eight minutes and swamped the audio with white noise on all tunes. I had to power cycle the C64 to clear it. Haven't been able to reproduce it since, so not sure if the problem is related or just an anomaly.

closed 2024-08-18 by frntc