frntc/SIDKick-pico #47: No audio after update from 0.14 to 0.20

crispycritter911, 2024-05-18:
I am using v1 hardware with an original raspberry pico from adafruit. I hand populated the board and added a DAC board.  I also modified the AV socket to do stereo out and everything was confirmed working on v0.14. On version 0.20, when I load "sys 54333" and hit the p key. I do not hear the test track as I do in 0.14. I've also tried SKPico Patch to add a few PRG files with audio and have audio silence as well. I've reverted back to 0.14 and it works correctly. This is on a 250407 board with a GAL PLA, a DIYChris RAM replacement RAM. The entire board was recapped.

frntc, 2024-05-18:
Are you sure you used the SKpico1*-firmware?

crispycritter911, 2024-05-20:
Yes. I used hardware version 1.

frntc, 2024-05-20:
... And the corresponding firmware? (Just to make sure we're on the same page :-))

frntc, 2024-05-24:
If the DAC board works with 0.14 there's no reason why it shouldn't with 0.20. I can't see any highlighted firmwares?

rittwage, 2024-05-25:
I have the same issue. In fact, the 0.20 menu goes crazy on mine and lines appear on the screen. I get no audio, or sometimes very distorted audio. 0.14 also works fine for me.

frntc, 2024-05-25:
@rittwage have you tried what I replied to you on forum64.de?

rittwage, 2024-05-25:
I will give it a try and see. I did play with it again last night.

I can reset to default, save settings. No sound. After power cycle, settings seems saved except volumes are at 0%. Turning them up does not make a difference, saving or not.

crispycritter911, 2024-05-29:
Just to clarify what firmware I am using. I've tried the following:
Firmware HW Rev 0.1/SKpico1_DAC.uf2 - bd7d38b0ca6d462214cb0e035bacfb15e515796e
Firmware HW Rev 0.1/SKpico1_DAC_LED.uf2 - c3a258f6a15e6f11365bf386afcff1a27b7b80c3

I've also tried the nuke file for Pico before reaching out.

frntc, 2024-05-31:
@crispycritter911 are you using a NTSC machine? If yes, check out firmware 0.201 which should solve this issue (I'll close the issue for now, we can reopen it if there are still problems).

@rittwage please also try the new firmware.

closed 2024-05-31 by frntc

crispycritter911, 2024-06-08:
So I tried .202 and it has brough menu corruption and locks my Commodore 64. Sound is still not working. I've tried the nuke.uf2 and two different v1 firmware. One for DAC LED and one that was PWM+DAC. Both cause instability and a hard lock in about 20 to 30 seconds. I've reverted back to 0.14 and all is normal. Also when I can get into .202's menu. I've notice the second SID warns me of IO 1 and IO 2 potential conflict. I've currently have the two wires soldered to CPU pins 12 and  (A5 and A8). I have nothing running to the cartridge port.

frntc, 2024-06-08:
@crispycritter911 still don't know whether you're on PAL or NTSC?!

crispycritter911, 2024-06-08:
NTSC. 450207 board. Board was recapped, diy Chris' RAM  replacement. GAL
PLA replacement. Replace a bad kernel rom from a donor board. Original
black power brick with the Commodore Saver inline. V1 hardware with the
official pico from Adafruit.

rittwage, 2024-06-08:
For the records, my issues were solved with 0.202 (didn't try 0.201).  

NTSC 250425 board with original PLA.

Seems there is sensitive timing somewhere.

crispycritter911, 2024-06-14:
I've ordered a second pla chip. I was using a GAL PLA but now I am going to try the PLS100N replacement too. I'll look the schematic for my board revision and start checking other in line components. I did have to replace the RAM as stated earlier and one of the 4066 chips.