frntc/SIDKick-pico #49: PCBWay Link

c64manguy, 2024-05-26:
The link to PCBWay says revision .2 but the picture of the board in the link is revision .1 and the boards I was sent when ordering from the link say revision .1... What version is actually uploaded on PCBWay?

frntc, 2024-05-27:
The Gerbers have been updated to v0.2 on 2024/05/17, if you ordered thereafter but received v0.1 you might want to contact PCBWay. The photo still shows 0.1, but simply because I haven't updated it yet.

closed 2024-05-27 by frntc

frntc, 2024-05-27:
There are changes on the PCB, but not fundamental ones. HW 0.2 allows for an additional way of reading the mouse/paddles, but the robustness also increased for HW 0.1 with the current firmware. In short, you probably won't notice nor miss the changes. (otherwise I would not have updated the project, but created a new one)