frntc/SIDKick-pico #50: V.2 Firmware 6581 Issue

c64manguy, 2024-05-26:
I am using a sidkick with the V.2 firmware (I think V.1 hardware, see my last issue post about that) with an NTSC 250407 C64 and when I try to change the SID type to 6581 and save it, the audio stops working. Then when I go back into the menu, there is corruption in the menu items and the volume is turned down all the way. If I turn the volume back up from the menu, nothing changes, and it still doesn't output any audio...
Is this the possible bug that you mentioned in the V.2 firmware download section? Or is it something new?
I have attached an image showing the menu corruption.
[image]

frntc, 2024-05-31:
Please check out firmware 0.201 -- it should solve this issue (I'll close the issue for now, we can reopen it if there are still problems).

closed 2024-05-31 by frntc

c64manguy, 2024-05-31:
Awesome. Thanks for all of your hard work!