frntc/SIDKick-pico #55: SidKick Pico in Eurorack

stilltryin, 2024-06-09:
Hi,

trying to use you module in another Eurorack module: https://github.com/busycircuits/alm012-sid-guts-deluxe gut it all up and running, but except a weird high pitch and bleep when i turn it off i get no sound...any ideas where to start?

frntc, 2024-06-10:
Not sure if I understood correctly: you hear glitches? Maybe the signal timing is a bit different from a real C64/C128 (for which the current firmware is tailored). What do you mean with "when you turn it off you get no sound"? (I don't have such a thing to test myself...)

stilltryin, 2024-06-10:
Basically if i get it right my module should work with a swimSID or OG and the sidKICK is a drop in replacement, so question would be it should work out of the box ?

frntc, 2024-06-10:
Not necessarily, as the SKpico has bus timings which work in the C64 and C128, other platforms might need different timings. You can try to set both bus timing values to "2" (or other values, might need experimentation) using SKpicopatch.

stilltryin, 2024-06-10:
will try thx

closed 2024-06-10 by stilltryin