frntc/SIDKick-pico #56: Slight X drift reading Y and vice versa on paddles

Fuzzweed, 2024-06-09:
Hi, latest revision is MUCH better for pot readings. However performing a full rotation of Y and it starts to affect the X axis readings too at lower values (and vice versa). On the previous firmware X and Y were all over the place with almost any X rotation causing Y movement as well, so I'm thinking / hoping its a software issue. I can't see any hard links between the pot caps on the C64 schematic and they're testing 1.5Mohm apart on the actual board, so don't think its hardware or 'normal' as such.

frntc, 2024-06-09:
It's probably the [VREF-issue](https://github.com/frntc/SIDKick-pico) (I assume you're using a Pico clone!?)

Fuzzweed, 2024-06-09:
Picos are genuine rpi, but I did put on the RGB firmware (didn't know if I had LEDs on the Pico tbh, so I thought I'd try....). I'll load up the vanilla firmware and try it

Fuzzweed, 2024-06-09:
No LED firmware is significantly worse. Getting mixed XY readings through entire range on physical pot X, and nothing on physical POT Y. I have two rev0.1 boards exhibiting same behaviour.
LED firmware same results as RGB firmware - reasonably accurate until value =~1E then some interaction between X and Y

Fuzzweed, 2024-06-09:
update: LED firmware has stabilised, just by being left alone for a few minutes. Some initialisation thing going on????

Fuzzweed, 2024-06-09:
update 2: X pot behaving pretty well, but Y pot maxes at and becomes jittery around ~DX (precise value not readable). Won't go to FF

frntc, 2024-06-10:
The firmware cannot stabilize over time, it's constantly doing them same thing. It sounds like something in your hardware part to me. The behavior you describe sounds like the VREF-issue, maybe there is a problem with the Pico-board. Maybe also the 4066 in the C64? What resistor value range do the paddles use?

Fuzzweed, 2024-06-10:
I have two sidekicks built at the moment doing the same so I think I can rule out the Pico for now, unless there a common build problem on the SIDkick. Are there any particular components / pins I can scope for clues or reference? The 4066 is a possibility, I have another 64 so I will do some tests on that and let you know either way.

closed 2024-06-10 by Fuzzweed