frntc/SIDKick-pico #57: RAD - Hardware Version 1 Firmware Version .202

steswart, 2024-06-10:
I am not getting intro/title screen music when my RAD is plugged in. Rad has latest firmware installed.

frntc, 2024-06-11:
Hm, works here. When you show the help screen ('H') what does the bottom line show as sound output?

steswart, 2024-06-11:
I just remembered that the title music is sampled, so I added the Digiboost option on the SidKick Pico config screen and now I can hear the music. 
BTW, when I press H on the RAD it shows:
 **RPi0 34C, C64/PAL, SID-type n/a**

frntc, 2024-06-11:
Interesting... the SID-type should read "SKpico" and then the RAD menu would use an 8-Bit DAC output mode which is independent of the digiboost. IIRC the RAD runs this detection once per boot of the RPi, does it always fail?

steswart, 2024-06-11:
Nope, rebooted a couple of times, re-setted the RAD, but it always shows SID-type n/a.
However, everything seems to be working fine now. All my programs on the RAD seems to be loading as before.
So, this issue can be closed. :-)

PS. Awesome projects of yours. :-)

closed 2024-10-06 by steswart