frntc/SIDKick-pico #62: Still trying to pin down pot X Y instability

Fuzzweed, 2024-06-16:
Following further testing from issue #56 replacing multiplexer, various caps and full continuity tests through the relevant areas I'm still getting interference between x and y readings.
Where I'm at:
SidKick works fine in 64C - but that's got my real SID in it normally so loath to do much comparative testing as I don't want to keep pulling the SID.
The bread bin with the problem: 
if I short (either) X to ground then I get perfect Y readings
if I short (either) Y to ground I get perfect X readings.

With X floating Y is read but is skittish and X readings start to drift too
"" vice versa

With X held high Y gets some very small movement or none at all
"" vice versa

As far as I can see from the schematics X and Y are electrically separate until they reach the SID, so despite fitting a new multiplexer anyway, I can't really see how this or anything external to the 'SID' is causing the issue. It's bit of a head scratcher.

I note there's a cap value change between from 1800pf on my breadbin to (I think the schematics are pretty sketchy) 470pf  on later machines - so I wonder if there's something going on in the discharge timings maybe? And there's some internal link in the pico/sidkick that's correctly resetting X when Y is grounded, and might also explain why the 64C is OK.....

I've got one of those cheap but also reasonable digital scopes if there's anything you can point me to to look at....

Fuzzweed, 2024-06-16:
additional note - I have rev0.1 boards - I noticed that rev2 firmware has additional trigger option for pot input, so is this possibly a known issue that received a hardware fix already?

frntc, 2024-06-16:
There is basically nothing that can go wrong on the SIDKick-side here, it's two plain digital GPIOs. If you tested in the config tool, then there's also no switching of the 4066. And on the firmware side I also can't imagine anything leading to that problem (you configured RAW mode for theses tests, correct?). Have you checked if the CBTD3861 is ok? (not sure how this could manifest, at least you could measure resistance between the Pot-X and -Y pins. Maybe some oscilloscope measurements on the POT-pins reveals something (also using a mouse)

Fuzzweed, 2024-06-17:
Thank you for your thoughts and time. I'll run through those. Any success I'll report back for reference otherwise I think I'll just chalk it up is an unresolved oddity.

closed 2024-06-17 by Fuzzweed

Fuzzweed, 2024-06-23:
Just adding a SOLVED in-case anyone else has this issue. Looks like there was a 'factory fitted' bridge between pins 23/24 of the SID. I think somewhere around pins 2/3 of the multiplexer as this is where traces are closest, but really impossible to see anything. Continually measured 300-800K between the two pot lines, even stripped back to a bare PCB mainboard, so not a hard short. Several rounds of wicking back solder to almost nothing and cleaning in that area seems to have fixed the problem.