frntc/SIDKick-pico #63: Audio "bumps" when all voices become silent, then again start to make audio

ZorroDJ, 2024-06-20:
I've tested all the possible settings, but it appears that a sort of "squelch" without a proper zero-crossing or DC-bias problem exists when emulated SID pass from no audio to some audio (tested with Mario Bros C64 port & Mojo Demo).
This causes a intelligible "bumb" on my speakers every time. Original sid is ok. Tested with PWM with no D/A.

frntc, 2024-06-20:
Yes, this is known (I implemented this to avoid a "click" when PWM is initialized and then used). Will be fixed at some point. It's not present in the DAC-output.

frntc, 2024-08-19:
This problem (except for an unavoidable small click on reset) should be gone with the updated firmware.

closed 2024-08-19 by frntc