frntc/SIDKick-pico #65: No sound on PWM using C64 250407 with swedish keys and swedish kernal.

mwedmark, 2024-07-28:
DAC works and gives sound but PWM gives a superlow sound (close to nothing) into TV via RF-signal.
Tried 3 different SIDkick pico with both later FW 0.2 and former 0.14. No changes.
Works with original SID-chip and gives sound to TV via original SID chip.

frntc, 2024-07-29:
For sure there's no relation to the Swedish kernal or keyboard. The original SID has different voltage levels, so you might want to check the circuitry between "audio out" at the SID socket and the audio/video connector (see https://www.zimmers.net/anonftp/pub/cbm/schematics/computers/c64/251138-1of2-combined.gif -- R6, FB13, Q8, R38, ...)

mwedmark, 2024-08-03:
Thanks. Yes there was a problem with a cable, not connected to SIDkick. I'm closing this as a non-Sidkick Pico problem.

closed 2024-08-03 by mwedmark