frntc/SIDKick-pico #68: Fred Grey's Shadowfire sounds very off from the beginning.

Muzzy011, 2024-08-09:
Fred Grey's Shadowfire sounds very off, I noticed that playing 1001 Crew Cracktro https://youtu.be/p6j-hlA6Wl4?si=b_qNcOlxAsH63q1a 

It sounds the same through PSID.

Frankie Goes To Hollywood from the same author sounds off too.

I tried it with the internal DAC, but didn't test it with an add-on DAC.

The same happened on the 2nd SIDKick Pico I assembled. Also, the same results were on two different C64, 250407 and 250425 boards. The same results with two different firmware, 0.202 and 0.14.

On the same C64 I tried it with the original PAL 6581 and SidKick Pico, through the same audio output (S-Video to HDMI converter, audio through HDMI to Monitor Speakers.

frntc, 2024-08-09:
Do you have a recording of your sound? Have you tried if your C64s need adjusted bus timing values? (via SKpicopatch)

Muzzy011, 2024-08-09:
I will record it later today and post it, with the original timings and adjusted ones.

I haven't tried adjusting bus timings before. Although I created firmware with 16 games in it, and saw those 2 params, 15 and 11. 
First is reading, second is writing times, right? 

I noticed that the same passages of the tunes will sound different depending on which point I started them, meaning the passage might sound correct if the tune is already played for 20 seconds, but if I started it right where it is, it will glitch out.

frntc, 2024-08-09:
You can leave your PRGs in. Try parameters with are a bit smaller and larger than the defaults (e.g. 13/9, 11/8, 17/13, 19/15, ...

The very old tunes rely on values floating on the bus, that might be one reason why it glitches differently.

modeler, 2024-08-16:
Fred Gray's driver does seem to be the acid test for these bus timings. I found that different boards seem more susceptible to these timings being off, e.g. breadbins / long boards are lot more sensitive than short boards.

I spent a long time testing out different values, eventually settling on this:

`SKpicopatch [path to .UF2 file] 05 07`

End result was really good, Fred's tunes (_West Bank_ and _Mutants_ were my test subjects) played perfectly in all of the boards I tested.

Muzzy011, 2024-08-18:
Modeler, thanks for the comment! I found pretty much similar results.

I did testing on 250407; what worked for me was either 06 06 or 06 05. Using 07 XX I could still hear glitches here and there, you can check the attached audio files (not very high quality, recorded with a cell phone from the monitor's speaker). I've attached all the tests from 15 11 down to 06 05, and you can actually hear the progression. The Breadbin board that I tested it on is a bit of a Frankenstein, as it had a lot of problems, so it has 8500 CPU, GAL PLA,  8565R2 VIC-II with 5V adapter, replaced Basic ROM with W27C512 EEPROM, and SIDkick Pico.

Frntc, I would say timing parameters solved the problem completely, just as a recommendation, please add one sentence in the documentation regarding this issue, with a brief explanation, that would be perfect!

Thanks a lot for reacting to this immediately, it took me some time to test and post the results here.

https://github.com/user-attachments/assets/e69fa4ab-3a17-458f-86a6-913b99f3cc37

https://github.com/user-attachments/assets/47d9b69d-f60a-4421-90df-b610b4ac040e

https://github.com/user-attachments/assets/dff896aa-2664-448c-841b-94cf31fd38e1

https://github.com/user-attachments/assets/bc01c237-a6b4-4c7b-9202-be79b10499cd

https://github.com/user-attachments/assets/e1300c8e-c16d-427e-9674-b28321d70d4b

https://github.com/user-attachments/assets/fb4930fd-45b6-48b9-9875-eb2b84941cc2

https://github.com/user-attachments/assets/345b207e-b853-4917-91ef-8e2c69ae05df

https://github.com/user-attachments/assets/4efff883-9062-4846-8cca-4b1f32043228

https://github.com/user-attachments/assets/7442dfaf-cdcf-451c-ba5f-393eb8afb186

https://github.com/user-attachments/assets/aea7b0c4-b081-4f46-be7c-d31c4b9ad350

frntc, 2024-08-18:
It has already been mentioned under https://github.com/frntc/SIDKick-pico#troubleshooting ;-)

closed 2024-08-18 by frntc

modeler, 2024-09-04:
> I did testing on 250407; what worked for me was either 06 06 or 06 05.

With the latest firmware (SKpico_FW0.21) I found I got some bus communication errors (text messed up in the file menu and such) with my previous settings, but everything worked perfectly with these:

`SKpicopatch [path to .UF2 file] 06 07`