frntc/SIDKick-pico #77: No Audio

THE-MrMicro, 2024-09-17:
Hi, first and foremost, Thank you! Your hard work and dedication are much appreciated.

I have assembled a SidKick Pico v.2 w/on board DAC. I have not connected anything to the DAC at this point. I am running this on a 250407 NTSC Breadbin and have tried FW 202 and 0.21for the .2 board and the straight PWM option, but I cannot seem to get audio out of this.

I can boot into the config screen and have disabled the second SID and tried SID 1 with an 8580 and 6581 with no audio by pressing 'p' OR by running the diagnostic cart.

Using a real 6581 produces audio.

Any thoughts?

Thanx much!

Best,
J

frntc, 2024-09-18:
Hey,

you have to use the DAC firmware and set the solderjumpers (https://github.com/frntc/SIDKick-pico#interface-board-with-onboard-dac), then you should get sound!

THE-MrMicro, 2024-09-19:
Ahhh, Ok.. I missed that whole section on adding the jumper. 

That and switching the FW got me sorted. Ty!

Will close this out!

closed 2024-09-19 by THE-MrMicro