frntc/SIDKick-pico #90: Pico don't save settings and corruption issue

Pinks64, 2024-11-02:
Hi I just bought and assembly Sidkick pico from 
https://restore-store.de/home/687-sidkick-pico-v02-2024-frenetic-4000000006879.html I also buy DAC but for now I'm not using it. My problem is that pico don't save settings and sometimes with some settings music is garbled. Also when I play test music from pico it's loud even sliders are on minimum/off position, after I change Sid there is no sound but after moving slider it works. Also now don't know why when I change Sid boost music is playing without interruption, earlier there was small interruption after each change. My mainboard is 250469 no. 252311 rev.b on other mainboard I have the same issue both pal.
[image]
[image]
[image]
[image]

This project is awesome, probably i just buy another kit and tray again or buy official pico and see.

frntc, 2024-11-03:
The configuration menu looks pretty weird. Have you tried different timings? Maybe the self-diag doesn't catch that one. See https://github.com/frntc/SIDKick-pico#troubleshooting

Testing with a different Pico is also a good idea, the black clones sometimes have quality issues.

Pinks64, 2024-11-04:
> The configuration menu looks pretty weird. Have you tried different timings? Maybe the self-diag doesn't catch that one. See https://github.com/frntc/SIDKick-pico#troubleshooting
> 
> Testing with a different Pico is also a good idea, the black clones sometimes have quality issues.

Thank you for replay, I will check this SKpicotiming tool as it can be run on C64 and I don't have knowledge of scripts :( also just recall that I have some dead test cartridge without harness and check that also. Yesterday I ordered official pi pico so 2-3 day and also check that one. Mouse works fine as I can tell.

Apologize for my English and again thank you for your time and project.

Pinks64, 2024-11-16:
> The configuration menu looks pretty weird. Have you tried different timings? Maybe the self-diag doesn't catch that one. See https://github.com/frntc/SIDKick-pico#troubleshooting
> 
> Testing with a different Pico is also a good idea, the black clones sometimes have quality issues.

Hello :) I have changed to official pico and now it save settings and show firmware :) 
I still have some glitches but there is another topic for this issue.
[image]
[image]
[image]
[image]

frntc, 2024-11-16:
The moved "pico" sprite and "A568IO" are just showing mouse activity (none attached) and detected additional address lines (if you meant this with glitches)

closed 2024-12-14 by frntc