frntc/SIDKick-pico #91: Program crash with the game HIGHNOON with new highscore table (by forum64)

TMO2018, 2024-11-17:
With my Sidkick pico rev. 0.2 (Interface board with onboard DAC) the game Highnoon (subsequent highscore list in Forum64) crashes during the opening credits in the picture of the duel. Firmware Revision 0.21 DAC is installed. Older and other revisions of the 0.2x behave identically. The opening credits work with an ARMSID.

Here is the link to the game:
https://csdb.dk/release/?id=247534

I suspect an incompatibility of the firmware.

frntc, 2024-11-17:
I just gave it a try with different SKpicos and settings, I can't reproduce the problem, works flawlessly here. I also didn't spot anything suspicious in the SID accesses.

TMO2018, 2024-11-17:
Thank you for your feedback. Unfortunately, I do not have a second SKpico for counter-testing. I will test the SKpico in another board.

TMO2018, 2024-11-17:
I tested the SKpico in a 469 board. There the opening credits work as expected.

In my 407 board, the opening credits crash. Maybe it's a sensitive board.

frntc, 2024-11-18:
You could try different timings (e.g. via SKpicotimings), but that's just a guess because the SID routine really seems to be non-critical.

TMO2018, 2024-11-18:
I have changed the two values in the range 8-13/10-16 with the program skpicotimings. However, no improvement could be achieved. The game crashes sooner or later in the intro.

I will use the SKpico in the C64 with the board 469. That's how it works.

Thanks again for your support.

frntc, 2024-11-18:
I'd be happier if the problem would be found :-) Some people reported smaller timing values, not sure if this might help. No two C64 are the same, as we all know ;-)

closed 2024-11-18 by frntc

TMO2018, 2024-11-19:
I found the problem.
After the intro crashed with the ARMSID and also an original 6581, I took a closer look at the SID socket. Some of the pins made a slightly wobbly impression. I then changed the socket. Now all SIDs work with the intro. Also the SIDkick pico. Thank you again for your support.