frntc/SIDKick-pico #94: Jupiter Lander cart missing sound effects

TMS2764, 2025-01-11:
I have a SIDKick pico and a SIDKick pico 2040DAC with firmware 0.21. The SIDKicks are working well on several PAL & NTSC machines but the Commodore Jupiter Lander cart has missing some sound effects.

frntc, 2025-01-12:
With the 2040DAC is this only after first power-up of the C64? (= does it work after pressing a RESET-button?)

TMS2764, 2025-01-12:
Same behavior on power up or after a reset. The main thruster sound effect works but all the rest don't. Start and ending tunes, side thrusters, fuel sound effects don't play.

frntc, 2025-01-22:
I'll have a look, but it will take a while...

TMS2764, 2025-01-25:
I'm in no hurry. Just trying to help. What I can say is that Jupiter Lander does work correctly once in a while. Maybe because I started another game previously. Its hard to reproduce and seems random. Once the game loads correctly all sound effects works fine. But if I load another game or reset then it behaves as before. I've never seen it work correctly after a power-up.

CBMD9090, 2025-03-03:
Hi, Awesome project. Thank you for creating it.
I have no Sound at all with Jupiter Lander - genuine CBM cartridge. And menu corruption.
I have built 5 SIDKICKs. I am using Pico clone RP2040+4MB, with SIDKICK V0.2 PCBWay.
so I checked SIDKICK PCB against the BOM. My PCB's are all missing R5 100K and also has NO solder short on Jumper (Hard reset).
Could this be why I am having problems?

frntc, 2025-03-03:
The Jupiter Lander problem is solved and there will be an update soon.

@CBMD9090 your problem is first of all (and most likely) that you need to adjust the bus timings (see trouble shooting). R5 is not really needed, the hard reset jumper is just a legacy thing.

frntc, 2025-03-03:
Firmware version 0.211 includes the fix for Jupiter Lander (and maybe other old ultimax cartridges), thanks @TMS2764 for reporting!

closed 2025-03-03 by frntc