frntc/SIDKick-pico #98: Menu corrupted even with timing correctio?

timtaylorretro, 2025-02-23:
First of all thank you for this great project!! 

I am having problems with several pico clones.  But it depends on the firmware version i use. 
See the pictures attached. When i have menu corruption with version 0.21 and use the older 0.14 version the menu corruption is gone. 

[image]
[image]
[image]
[image]
[image]
[image]

frntc, 2025-02-23:
The data (what is stored in flash memory and how it's used) has indeed changed since 0.14. However, not spotting an error in the menu does not necessarily mean that all works correctly. The components on the pictures seem to be fine (same flash, rp2040 stepping etc.), but what happens from time to time that these Pico clones suffer from quality problems. I had some with weird boot problems, maybe it's just bad soldering or routing. I think the best advice I can give is the change the Pico clone... (if you want the new firmware)

TMS2764, 2025-03-08:
I believe I am experiencing the same issue. However my Pi Pico is not a clone. I bought it from Mouser and this SIDKick Pico was working before I updated to 0.211. I have updated other SIDKick Pico and SKpico2040DAC to 0.211 without issue. However this with this SKpico in particular I was experimenting by doing a special build by adding .prg files to the .uf2 file using SKpicopatch. Since then I got this menu corruption. I have since reprogrammed the SKpico with the unaltered .uf2 file but I get the same menu corruption. What is funny is that the SKpico works fine. All sounds play properly. There is no bus timing error. The menu shows garbage but it does work. I can change settings and it works.

LouDnl, 2025-03-08:
Take the Pico out of your C64 and flash it with the uf2 in the link below. It erases the flash and then boots in dfu again so you can flash the skpico firmware. 
Maybe it helps, maybe it doesnt.
https://github.com/dwelch67/raspberrypi-pico/blob/main/flash_nuke.uf2

TMS2764, 2025-03-08:
Its a good suggestion. I tried it but unfortunately it didn't change anything. The menu is still showing garbage but other than that it seems to work fine.

frntc, 2025-03-08:
> I believe I am experiencing the same issue. However my Pi Pico is not a clone. I bought it from Mouser and this SIDKick Pico was working before I updated to 0.211. I have updated other SIDKick Pico [...]

can you send me the .uf2 that "corrupted" your Pico? (best to my email address, it's in the source code files)

CBMD9090, 2025-04-07:
Someone has identified a common cause of Menu corruption, which is not caused by bus timing or soldering. Lots of Great info here, if you have built your own SIDkick Pi Pico and are having problems:
http://spacetaxi.epizy.com/MAIN/SIDKick.htm

frntc, 2025-06-10:
I'll close this thread, it seems the issues have been resolved.

closed 2025-06-10 by frntc