frntc/SIDKick-pico #111: Disappearing sound in Tenebra & Tenebra 2

keronian, 2025-05-21:
I'm having issues with the sound specifically in Tenebra and Tenebra 2... there are some footstep sounds that are supposed to play as the player walks around, and while everything STARTS fine, it fades quickly into nothingness.  If you keep playing, at points it does pop back up again, but it does the same thing.

The sound at the beginning of Tenebra 1 also doesn't seem to play at all until you start the game, at which point it makes some odd sounds...

My guess is in both cases the game is doing some nonstandard things with the audio, but if you could take a look it'd be appreciated.

keronian, 2025-05-22:
PWM.  Basic configuration.

frntc, 2025-05-23:
Maybe it's an effect of the "anti-click-noise" behavior for the PWM output. Do you have a chance to test with DAC? (of course I could do that as well, but it'll take a while)

keronian, 2025-05-23:
I do not, sorry.

frntc, 2025-10-04:
I finally had a bit of time to look at this. It's an interesting side effect of the 6502's (+family) indirect addressing in Tenebra's speed code that puzzles the SKpico. A fix will be part of the next firmware version, hopefully coming soon together with some other updates.

keronian, 2025-12-30:
It does seem to have fixed the issue, thanks!  The opening screen still sounds a little weird on Tenebra, but that's probably weird analog SID things, so I'm pretty happy. :)