frntc/SIDKick-pico #113: What is the magic timing?

RevivingRetro, 2025-06-04:
I have a dozen v2 DAC boards - all came assembled from PCBWay. I am using Amazon purchased Raspberry Pi Picos (vendor Vis Viva). 
I have tried a multitude of different timing settings 13/9, 12/8, 6/6, 5/5 - using their own unique assemblies and picos and different C64 boards (250407 and 250425). While the 12/8 and 13/9 timings seem to fare much better than the default 15/11 - there is still sporadic no sound and access to the control panel via sys 54333.  I have to reboot multiple times to get either to work.

Am I missing something here?

[image]

RevivingRetro, 2025-06-05:
Thank you for the reply. Interesting you had to use 100 for the padding. I used 20 and pass the harness test with no error.
I followed the thread you linked and it seems like its going to take a lot of experimenting. I don't want to give someone a machine with these types of issues though, so I'm going to try the simple 2040 and if that doesn't work out, I'll go back to the legacy.
These are only costing around $12usd in parts/build so it's not much an investment to walk away from. :)

SX64man, 2025-06-05:
Phi2 the first parameter is easily viewable on 6510/39 but not sure where to get the readbus value for the best troubleshooting of exactly what does the second parameter change. A typical scope view of phi2 (yellow), and a5 (purple) is below for a working combo with good audio, menu, etc.. 
Unfortunately the non-working scope version is not really any different using those same inspection points.

[image]

SX64man, 2025-06-06:
On passing the diags - I noticed something interesting.  The diag always passes for me when the firmware is first loaded and the config is in default mono mode.  However when I SYS in there and change it to my preferred stereo setup combo (see below) then that's where it no longer passes and I must reflash with the "100" value 3rd parameter to get the diag to pass again.  

[image]

SX64man, 2025-06-06:
This is the hardware setup I have been using on the 5 machines.  Rather than getting a5/a8 directly from the 6510 - I get them at the closest vias carrying the signals.  Here is a 250407 NTSC board.  I have also tested getting a5/a8 at the 6510 and it made no difference in behaviour in terms of audio or menu corruption/lockout etc.

[image]

..

SX64man, 2025-06-06:
Excellent good to hear!  I think the difference with the diag is probably due to the long cables from the user port to the joystick ports on my harness. Looks like yours is a more local design at each port and probably has less loss to the joystick ports.

SX64man, 2025-06-06:
I also did some more testing using the SKpicotiming.prg and was was interesting was for most of my NTSC machines, 11/7,12/8, or 13/9 seemed to be the sweet spot as soon as it got to 14/10 boom the dreaded unstable message popped right up.  I think I will suggest to FRNTC's documentation that for NTSC the default might be safer as one of the 3 values above rather than 15/11. I'm curious if other NTSC users have the same experience.

frntc, 2025-06-06:
Great that you figured out values that work. For the signals: Ax comes from the CPU and is typically less of an issue -- more interestind is the CS. And yes, in general the clock frequency is slightly higher for NTSC and thus smaller values are likely to work better. The ideal paddle value mostly depends on some capacitors on the C64 board, parts which do not get better over decades...

What sporadic glitch when hovering over pot-values do you mean? The "pico"-sprites move according to the paddle values, is it that?

SX64man, 2025-06-06:
Thanks for the explanation.  Absolutely. great work!   I'm going to do some scoping of CS and see if it will be possible to dial in the timing values for the individual machines.  Of my 5 NTSC 250407s,   3 are using 11/7/100, 1 using 13/9/100 and the last one is a Bwack 407 reproduction mb and is using 10/6/100.  I have had these 4 vintage machines repaired for a while but no SIDs to put in as the original ones had all died.    So your solution is perfect, in particular because it can do stereo very easily. 

On an unrelated subject, maybe should be on separate thread.  Is it possible to tweak the 6581 hi pass filter so that less artifacting is present or is this a limitation of the hardware not being able to sample the higher frequencies cleanly.  It's pretty apparent in 6581 mode using SidTester filter test.  In 8580 mode it seems better. So perhaps it's something buried in the original reSid imalgorithms. algorithms.  it's not a big deal butvthatvus the reason I am using dual 8580s with digital in the config.
awesome work!

SX64man, 2025-06-06:
Did a few more tests today - I wanted to test settings on a modern NTSC C64 (https://github.com/bwack/C64-250407-Replica-KiCad) board with a few mods and modern replacement ICs.  I'm happy to report that this 11/7/100 setting produces good stereo audio and passes the diag tests on a modern mb with new caps, components, JCIAs, Gal PLA, TEBL modulator replacement customized for DAC (no internal audio), 6510 with Phiagra buffer,  and motherboard taps for a5/a8. 

[image]

SX64man, 2025-06-06:
scope tests on differences between 11/7/100 and 13/9/100 don't appear significant to establish a clear difference in timing shift.   Differences, if any, are very subtle.  

Measured at SID 6 (phi2) and SID 8 (CS)

[image]
[image]

RevivingRetro, 2025-06-06:
@SX64man

Did you try diag with a 20 offset instead of 100 with your latest setup? 
Curious to see if this does indeed boil down to the harness you are using compared to mine since offset of 20 passes diag tests with my harness.

Does the offset of 100 compared to 20 have significant difference in the actual use case? 
It seems like a large disparity from the default.

SX64man, 2025-06-06:
Interesting experiment! - yes, on the new bwack board with the new caps, the diag passes with 11/7/20 as well . So that would confirm @frntc post above about the caps - none of the vintage 407 boards that I was testing with have been recapped.  My understanding is that the 20 parameter is an offset, but the 100 is a hard set parameter.

RevivingRetro, 2025-06-06:
Thanks for checking the paddle responses. Now that I see the paddles stay within range, it seems to me that going to 100 offset might be better than staying with the 20 in order to account for older machines that haven't been recapped. :)

SX64man, 2025-06-07:
Ill repeat that experiment a few times over the next few days - I actually expected the values to be identical but they are pretty close and it could be just an arbitrary "moment in time" difference.

VanessaE, 2025-08-03:
I'm trying to get my SKPico 2040DAC going on my NTSC C128-DCR...

After a bunch of power cycling and wiring confusion, I temporarily got at least to a point where it the SKPico was begrudgingly willing to respond.  Up to that point, it was acting as though it were completely dead, at least as far as the 128D is concerned.

Much like the OP, I've tried a range of readbus and phi2 values, going as low as 5 + 9 and 6 + 8, but so far I've not found a reliable setting.  Sometimes it works, but usually not.

At one point, I managed to get values of 6 and 10 into place, and the menu started behaving, and I was able to run it for about 20 or 30 minutes before the SKPico stopped working again. 

I was playing music during that interval, and it worked, if unreliably -- the music was glitchy (dropped notes, off-key notes, extra sounds, inconsistent volume), and digi routines such as the "Mahoney method" would occasionally fail to initialize properly, but I was only testing to see how long it would go before failing.

Once it stopped working, it was back to the way it was initially, with maybe a 1 in 30 chance, if that, of even initializing itself (I judge this by listening for the "pop" when the C128 prints its banner and clears the SID).  Nevermind getting into the setup menu.

I eventually reset the timings back to the defaults of 11 and 15 when I got the opportunity, since I can't find any pattern to when it will or won't work, or really any meaning to these numbers for that matter.

My old dual SID board sporting a pair of old 6581's Just Works™.  I wanted to replace it with this SKPico because those are first-revision SIDs, i.e. really old, with a lot of analog background bus noise and mediocre filter behavior.

I'm totally lost.  I'm not above blaming the computer here, but I did try the SKPico in my other C128 and it didn't work there, either.  I even replaced the motherboard's SID socket with a nice new double-wipe part, and cleaned the SKPico's pins \*thoroughly* just to eliminate the possibility of a connection fault.

VanessaE, 2025-08-03:
> To avoid confusion, let me mention this here again: $d700 is neither checked nor tested. Checking an address range is nothing the SKpico can do on its own from the SID-socket

Well of course I mean the configuration tool here.  I'm assuming the 6510/8502 is running ordinary code at that point, and thus can look wherever it wants.

> I'm still not convinced that your SKpico is working flawlessly hardware-wise. 

Well, the vendor I bought it from is 100% certain of his build quality.  We went round and round in email trying to sort this out; he was initially focused on poor connection between the SKPico and the socket, but clearly that can't be the problem here, and he has since relented.  He thinks it's the same timing problem as discussed in this issue.

> Maybe you can get the PCBs for the cheaper version just for testing?

Unfortunately I can't really do that.  Apart from simply not having the eyesight or dexterity for this anymore (it's one thing to just look at something under a magnifier, and another to try to build something that way), I don't have any relevant parts; it definitely wouldn't be cheaper.

frntc, 2025-08-03:
Well, I can't judge from here if the SKpico-hardware is 100% ok. If you assume it is, I can only recommend to systematically test the timing values (some "NTSC-friendly" settings are mentioned above). Just taking the smallest ones is also not the best strategy. In any case there's enough room to accomodate timings, and from the feedback I got, it was typically some other problem (sockets, cold soldering joints, low quality signals such as Phi2, voltage levels, ...). Don't you have other C64s, PAL machines, ... so approach the problem step by step?

frntc, 2025-08-03:
The relevant bus communication and timing is starting here: 

https://github.com/frntc/SIDKick-pico/blob/main/Source/SKpico.c#L1105

> At one point, I managed to get values of 6 and 10 into place, and the menu started behaving, and I was able to run it for about 20 or 30 minutes before the SKPico stopped working again.

Did it play the SID tune in the config menu for some time and then stopped playing?

frntc, 2025-08-03:
All that is strange. Can you patch the firmware with some NTSC-friendly timings and flash is again via USB? Still not convinced that the SKpico hardware is in good shape.

VanessaE, 2025-08-03:
Sorry, I had to delete that last post.  I'm getting tired so I'm kinda rambling.  I'll repost in a few minutes, after rewriting it.

VanessaE, 2025-08-03:
Okay, let's try this again:

> The relevant bus communication and timing is starting here:
> 
> https://github.com/frntc/SIDKick-pico/blob/main/Source/SKpico.c#L1105

Noted.  I never thought to look at the source code.

> Did it play the SID tune in the config menu for some time and then stopped playing?

There's a tune?  I don't recall seeing any mention of one in the config menu and there's no obvious mention in the README.  In any case, I was using regular music players during my tests.

Curiously, the SKPico was somewhat behaving for a while there after taking that 'scope photo, and I didn't do anything at all to it since then -- in fact it, still had the 20/25 timings from the last tinkering.

By "somewhat" I mean that it suddenly decided that it should start up silent and then emit continuous noise from the left channel when the KERNAL/BASIC clears the SID.  Still, since that meant it was responding, I took a chance and was able to call-up the menu.  I found that tune you mentioned, in the "help" screen.  Peculiar place to mention it. 🙂

Since I was there, I played it, and it worked, albeit with the occasional bad note.

The menu complained of unstable timing, so after I finished listening (I let the tune repeat a few times), I took the opportunity to run the timing tool and set the timings back to something saner (6/6), and reset the machine.

Aaaaannnd, now it's gone back to just not working at all, which doesn't make any sense to me.

I did tweak the motherboard's clock trimmer between taking the 'scope photo and the above playing, but I was just trying to clear up some very mild smearing/misalignment of the chroma signal.  That had very little effect on anything anyway (the machine might need a new 8701, but the system clock is rock solid, so I doubt that's the problem).

Continuing, to take into account your most recent remarks:

> All that is strange. Can you patch the firmware with some NTSC-friendly timings and flash is again via USB?

Sure, I can do that.  What values should I try?

> Still not convinced that the SKpico hardware is in good shape.

Neither am I, but it could also be a fault in the way the SKPico communicates with the motherboard -- maybe you're taking a shortcut or making an assumption somewhere that only works on C64's and/or PAL machines?

SX64man, 2025-08-03:
You might try flashing - SKpicopatch firmware.xxx 15 11 100 or  15 11 20 and see if that gives you a more stable setup.  I would include the third value - the joystick port diag test offset.  Those values seem to work for my NTSC machines but they are all C64 and not 128 so YMMV.   I would keep the scope connected. Is your clock value closer to 1.0227  ?

VanessaE, 2025-08-03:
> You might try flashing - SKpicopatch firmware.xxx 15 11 100 or 15 11 20 and see if [...]

Aren't 15 + 11 the default timings?

> I would keep the scope connected. Is your clock value closer to 1.0227

It \*should\* be 1.0227, yes, but I don't have anything that can measure frequency other than my 'scope, and its accuracy in that regard is really poor.

SX64man, 2025-08-03:
sorry I misspoke 11/7/100 or 11/7/20

VanessaE, 2025-08-03:
Okay, flashed with 11/7/100 in place and took a count:

Over the course of 20 power cycles, the SKPico initialized well enough on 11 of those tries, such that I could at least get the system bell (by pressing ctrl-G).  In all of these successful starts, the bell sounded weird compared to the norm (it's more metallic, and a bit distorted).  In the remaining 9 of those 20 tries, the SKPico just remained silent.

I tried a couple more power cycles, looking for another successful startup (the last of the 20 was bad).  Once I got that, I pulled up the menu.  There were no errors or complaints of any kind.  I played the demo tune, and it seems fine I guess -- I don't know how this song should sound.

I've left it playing as I submit this comment.

VanessaE, 2025-08-03:
Interesting, after letting that tune play for a few minutes (on repeat), I power cycled the machine out of curiosity, and this time the system bell sounded clean, completely normal.

I have not changed the timings yet -- they're still at 11/7/100.  I'm still struggling to understand WHY.

SX64man, 2025-08-03:
I have no idea why either, but and interesting thing is that when I flash the firmware with the settings, I have better results than when I try to use the timing utility to do it (I have it as an entry in the menu).  I really dont know why it would be different - it shouldn't, except that the app runs  in terminal/cmd mode to flash as opposed to native on the c64/128.

VanessaE, 2025-08-03:
> when I try to use the timing utility PRG in 64 mode to do it

I wonder if theC64 timing tool is broken in some way?  Or perhaps it's just not compatible with the SKPico 2040DAC ?

>  Actually the app in 64 mode does not have an entry to change the third offset value (If I remember right), so it might have something to do with writing that third value to the firmware, even if you are not changing it

You're right, it doesn't allow changing that third parameter, but isn't that one only used for the Pot X/Y inputs?

SX64man, 2025-08-03:
exactly,  it shouldn't make any difference as it was just designed to pass the diag test.. but I dont know the code interaction to see how it accomplishes that.

SX64man, 2025-08-04:
@VanessaE Out of curiosity, do you get the same menu stability results with 11/7/100 in both C128 and Go64 mode ? and what clock frequency does it list on that config menu in both modes ?

VanessaE, 2025-08-04:
> @VanessaE Out of curiosity, do you get the same menu stability results with 11/7/100 in both C128 and Go64 mode ? and what clock frequency does it list on that config menu in both modes ?

"Non-sequitur, your facts are uncoordinated."  🙂

When you call up the menu from C128 mode, it just drops to C64 mode to run it.

That aside, it correctly reports NTSC, 1022727 Hz, but I'm gonna guess that that's just a display value, and not a measurement.

***

In a tangentially-related matter, I fixed the color fringing problem that I mentioned -- it was a dodgy 8564 VIC-IIe chip.  Replaced the 8701 also, though that turned out to be unnecessary.  Always good to have spares on-hand.

I only mention this because all of the system timing goes through the VIC-IIe and 8701, but replacing those chips had no effect on the SKPico's behavior, which tells me the system clocks and other critical timings were fine before (and still are); the 'scope was actually telling the truth.  🙂

VanessaE, 2025-08-04:
> SKpicotiming is working differently to avoid too much special code on the SKpico side. Be assured, it the communication works for most bytes sent, then it changes the timing settings. The test case is the config tool :-)

That's fair, but you did say it's impossible for the tool to be 100% reliable when the timing already is way off, and if that's the case, then it needs to verify what's been written.

Plus there seems to be a vague correlation between attempts to change the timings and the SKPico deciding to go incommunicado, but I can't be sure because to be honest, I was not being too careful there.  That can happen when one tries to troubleshoot while tired and sleep-deprived.  😉

> Assuming that the pin header and the socket connections are okay,

They definitely are okay.

Like I said, it's a brand new motherboard socket, and if I'd scrubbed the SKPico's pins any more, I'd probably have started to strip the plating off. 😉   No way is there a connection fault.

>  I still think your SKpico has a hardware problem

Well, I've turned the machine on and off a few times today and it hasn't failed to initialize at all, though I haven't played any music, just checked if the bell works.

That would seem to imply that the SKPico IS working...?

> I assume that you got it from one of the official vendors

Yes, it came from Retro8bitshop and appears to be machine-built, pin headers and LED aside.

That supplier insists that he's sold 200+ units without a single failure, so if there IS a hardware fault, then I must have somehow damaged my SKPico while installing it.  I would very much like to rule this out.  Damn thing cost enough already.

VanessaE, 2025-08-04:
> I know you understand what a ridiculous statement that is from the seller.

Yeah, I'm not willing to accept that claim at face value, but I'm also not above taking the blame myself, either.

rittwage, 2025-08-04:
These devices are extremely cheap. Just order another one from someone else
and see if you have the same problem.

People have a lot of trouble from bad RP2040's.

VanessaE, 2025-08-04:
> These devices are extremely cheap. Just order another one from someone else
and see if you have the same problem.

I wouldn't call 28 euros plus shipping, "extremely cheap"...  Can you suggest an alternative vendor, possibly someone in the US?

VanessaE, 2025-08-04:
I find no SIDKick Pico 2040DAC for sale on ebay, either as kits or assembled units.  Just the older sandwich-of-boards designs.

VanessaE, 2025-08-04:
> I guess you are beta testing. :)

This ain't the mission I signed-up for 😛

VanessaE, 2025-08-04:
> I don't remember if he also offers/plans to offer the 2040DAC

Indeed he does, that's how I acquired it:

https://www.retro8bitshop.com/product/sidkick-pico-2040-by-frntc/

> I prefer not covering the LED, but everyone can do as they please (and fits the mainboard).

So move the LED 😉

> Since I haven't heard of any fundamental problems, I'll try to pinpoint the error here (which I suspect is with this SKpico).

Yeah we're way off in the weeds now.  Better to steer this back on topic.

VanessaE, 2025-08-04:
So, I was thinking... would it be possible, using the timing adjustment tool as a starting point, to create a tool that can auto-tune the timings?

Here's how I could see it working:

1) Set a flag indicating which direction a change of timing value will go -- either up, or down.
2) change the timing value by 1 per that direction flag
2a) if we hit the limit of the timing value's range, back the value off and invert the direction flag.
3) verify for sure that the new value was accepted
4) do a whole bunch of read and write tests (say in the thousands)

5) measure the failure rate and add it to a table/array.
5a) If the failure rate was 0, Bail out of this part of the autotuner.

6) compare the failure rate now with the failure rate of the previous pass.
6a) If the failure rate was better, loop back to (2) for another pass.

7) if the failure rate was worse,
7a) if the program has already inverted the direction flag at least twice, this means we're stating to ping-pong, so scan the table of results, get the timing value for whichever entry was best, and bail out of this part of the autotuner.
7b) if the program has not yet inverted the direction at least twice, invert the direction now and loop back to (2) for another pass.

Sooner or later, either 5a or 7a will be satisfied.

Now, commit the final timing value, and do the whole thing all over again with the other timing value.   I.e. the first part of this hypothetical program could do the Phi2 value, then the second part would do the Readbus value.

If none of that is possible, how about just a tool that will do a bunch of read/write tests without changing timings, report the failure rate, and exit?

The idea here is to have something objective and measurable to refer to.

RevivingRetro, 2025-08-04:
@VanessaE

https://www.ebay.com/itm/277307932972

VanessaE, 2025-08-04:
@RevivingRetro that was a nice offer, even though you deleted it from this thread before I could finish my reply. 😉 

Offer accepted.  Please check your email.

In the meantime, I still would REALLY like some way to verify without a doubt, one way or the other, if my current SKPico is bad.

If the one I get from you also misbehaves, I'll just keep both anyway, as that would mean the problem is probably a firmware bug that @frntc just didn't run into on his machines (we all know how inconsistent Commodore manufacturing could be from one machine to another), and I had planned on buying another in the future anyway, to upgrade another machine.

frntc, 2025-08-04:
> Now, commit the final timing value, and do the whole thing all over again with the other timing value. I.e. the first part of this hypothetical program could do the Phi2 value, then the second part would do the Readbus value.

Unfortunately you cannot fix one value and then the other. For reading values from the SKpico you need both and readbus depends on the other (i.e. it's a "2D optimization"). Independent of that: there were reports about lower timings for NTSC machines, but finding the timings did not seem to be a huge problem so far... if it is a problem with the firmware, I'm happy to dig into it of course ;-)

VanessaE, 2025-08-04:
> For reading values from the SKpico you need both and readbus depends on the other

Well, nothing says you couldn't change both and do it all in one loop instead of two.  The same reasoning would apply, I would think -- the underlying idea is simply to see if a given pair of values is objectively better or worse, and ultimately, rendering this github issue obsolete. 😉

> but finding the timings did not seem to be a huge problem so far

...at least until I came along and tried to set one of these up on an NTSC C128-DCR. 🙂

RevivingRetro, 2025-08-04:
@VanessaE I'm sending it out now. I upped it to USPS Priority (vs Ground) so hopefully gets to you a little faster.

I'll break out one of my 128-DCR machines when time permits in the next couple of days and see if I have the same issues as you.

LOL on the headers :)

frntc, 2025-08-10:
https://github.com/frntc/SIDKick-pico#firmware-which-one-to-choose

https://github.com/frntc/SIDKick-pico#troubleshooting

VanessaE, 2025-08-10:
@frntc well, I meant, what more than this?  I already know which firmware, but considering the preceding conversation, I'm reluctant to just throw random timing values at it...

frntc, 2025-08-10:
I'd try the timings mentioned above -> https://github.com/frntc/SIDKick-pico/issues/113#issuecomment-2942488078

VanessaE, 2025-08-10:
I have tried the following timings:

13/9/100
11/7/100
10/6/100
9/5/100
8/5/100
7/5/100
6/5/100
5/5/100

None work.   No sound, no menu.

I'd try to go lower, but `skpicopatch` won't set timing values of less than 5.

Just to make sure I'm doing this right, this is my command line:

`rm SKpicoPRG.uf2; sleep 2; ./skpicopatch ../Firmware\ 2040DAC/SKpico2040DAC.uf2 13 9 100`

...and likewise for other timing values (the `sleep` is just so that I can see when `rm` deletes it).

Then I plug the the SKPico into USB and copy `SKpicoPRG.uf2` to it.  I let the SKPico reboot itself, and then I unplug USB and move it to the 128-DCR to test it, taking extreme care to fit and orient it correctly each time, and testing with no wires attached (per the README).

As before, a real SID chip works exactly as expected.

Now what?

frntc, 2025-08-10:
Can you send me the patchted .uf2? I'll give it a try here. (my mail address is in the source code files)

VanessaE, 2025-08-10:
Email sent.

RevivingRetro, 2025-08-10:
@VanessaE I didn't "forget" the LED nor configuring the amp pads. I sent you a freshly built board. I don't include the LED on purpose (I order in bulk and this increases cost for something not needed since I do not sell these to the public and include them in systems already configured). I thought you would configure the board and flash to your own specs. So sorry I wasn't clear on what I was sending you.

Considering that I should have been more clear in what you were purchasing and I should have asked for more detail, I processed a full refund for you, including shipping.

VanessaE, 2025-08-11:
@frntc 

> Maybe you made any modifications to your C128(s)? Any hardware attached? Did you run any diagnostics recently?

Well it does have some mods, but nothing that would interfere with addressing or a SID.  Just basic stuff like JiffyDOS, 1571 device number, a couple of auxillary power taps, and an RFOFF replacement.

It also has an internal serial connector for a µIEC/CF installed in the machine.

I plugged in the diagnostics dongles and ran through the tests just now.  Everything checks out, except that it reports cassette and serial as "bad".

I think those two are false positives, since it's a C128-DCR with integrated 1571, can't unplug

@RevivingRetro 
> Considering that I should have been more clear in what you were purchasing and I should have asked for more detail, I processed a full refund for you, including shipping.

You didn't need to go and do that...  It was easy enough to correct those jumper pads, and LED.  If you'll tell me how, I will return the refund back to you.

VanessaE, 2025-08-11:
Oh accidentally committed that message before I was done.  To finish:

I think cassette and serial are false positives, since it's a C128-DCR with integrated 1571, can't "unplug" that, and the cassette port does share one line (4/D with /SRQ, to CIA \# 1 /FLAG).  I guess this simply interferes with the loopback plug, as serial does work just fine in all modes.

The µIEC is not plugged in, so there's nothing there to interfere.

VanessaE, 2025-08-11:
@RevivingRetro 
> Yes, I have a scope. I have a few DCRs - including one with JiffyDOS like you seem to have so I'll do some testing when I get back.

I'd appreciate the comparison, though it looks like there wouldn't be a point, see below.

@frntc 
> not sure what you mean with "rather hot"

As in reaching close to 5 volts on the high side of the clock, though now it looks like that's not the issue either...

> It's reading Phi2 in a loop as I've done in all my other hardware projects as well.

Out of curiosity, is it going to a TTL/GPIO line, or via some analog input?

> While 3.7V is enough to be detected as high, it's definitely not the normal voltage level for a C128

Apparently, it is.  Here's a capture of 1 MHz, R/W, and SID /CS on one of my old "flat" 128's, where the KERNAL is initializing the SID at bootup:

[image]

As you can see, all three have substantially the same signal and slopes/curvature as the DCR's readings.  The machine had an old 6581 in it for that capture.

> maybe the signal is fluctuating (beyond the resolution of your scope)

I suppose that's possible, but wouldn't that break pretty much everything?

Curiously, the SKPico works on that old machine, at least for the first couple of tries.  Both menu and sound.  I think it's set for 11/7 timings (I forget, too many tests, too many confusing results).  I did not test it extensively since that machine isn't where I want the SKPico to end up.

What else can I check?  (note that I do not have access to a logic analyzer).

VanessaE, 2025-08-11:
I decided to take a look at the address and data buses (as they appear on the SKPico's pin headers).

Address lines all look clean, nothing unusual going on at all.

Data lines are okay, though sometimes a signal only reaches 3.7v or so for logic high, similar to the clock.  As I'm given to understand, this is pretty normal for a C64 or C128 data bus, since different chips will have different drive strengths.  Here's D0, for example:

[image]

(1 MHz on channel 1, SID /CS on channel 2 this time, and D0 on channel 3; trigger is on CS falling edge)

I just had the machine in a loop "printing" ctrl-G repeatedly to ring the system bell, to keep the SID /CS line active.  Since the trigger is the falling edge of /CS, I assume the CPU is driving D0 at that instant.

frntc, 2025-08-11:
I checked Phi2 on my flat NTSC-C128. In fact it's voltage level is even lower (and the SKpico still works), but I'll check the schematics to see why this might be the case. Did I get you right, it's working on your flat C128 now?

VanessaE, 2025-08-12:
@RevivingRetro 
> I turned the DCR on and off multiple time and successfully got the GUI each and every time.  I get sound just fine as well.

Thanks for the test.  Don't worry about the 'scope traces, guess it doesn't matter now.

@frntc 
> Did I get you right, it's working on your flat C128 now?

It works on the flat 128 I tried it on, but I only  gave is a quick check, menu and demo tune work, but that's as far as I went since that machine is third in line behind the DCR and my other flat 128.  I only used it because the case was already open. 🙂

Well I'm stumped.  Why would the SKPico not work in my DCR, but real SIDs work fine? What could possibly be the fault here?

VanessaE, 2025-08-12:
I should reiterate, it's not a consistent failure.  Every once in a great while, sound and menu both work.

For instance, I tried again just now, just scoping out the +5v and reset lines on the DCR to see if anything looked wrong there (they're clean and have a quick rise).  For no obvious reason, both menu and sound started working...for exactly one power cycle, then they were gone again.

RevivingRetro, 2025-08-13:
I gotchya - since you mentioned SuperCPU - on a side note, I have these SuperCPU adapters in several of my DCRs. A bit pricey, but they're cleaner than the original adapter - it also makes it easy for me to just plug the SuperCPU into any of them and go :)

https://www.ebay.com/itm/167672955508

Also interesting that you mention you were looking for clean stereo output - in my opinion, and my untrained ear, the SKPico doesn't feel like a cleaner sound than the 6581s I've used - maybe they're later revs. I like the SKPico because it's the good value and offers diverse flexibility. I use a lot of them in the systems I've configured.

Have you looked at the ARM2SID +FM?

I really hope you ultimately get this figured out.

VanessaE, 2025-08-13:
@RevivingRetro 
> I have these SuperCPU adapters in several of my DCRs. A bit pricey,

Eeesh, you're not kidding.  I at least  modified my adapter to have the wires connect to it by a proper header, and replaced the hook clips that CMD supplied with much smaller, more compact clips.

> the SKPico doesn't feel like a cleaner sound than the 6581s I've used - maybe they're later revs

Well it kinda depends on what you're doing.  It's this sort of whine with white noise overlaid.  Every 6581 I have ever used on every machine I've ever had has exhibited this noise to varying degrees; newer chips are better, and the one 8580 I have is quite clean.

> Have you looked at the ARM2SID +FM?

I have considered that one, but the maker has already stated that he won't release firmware source code and design files until after he ceases commercial production.

> I really hope you ultimately get this figured out.

I'll keep tinkering.  I almost had it going earlier today, and the successes vs. fails gave me a direction to look in at least... I'll post details when I finally solve it.

[21 more replies]