frntc/SIDKick-pico #118: troubleshooting advice

pentode1, 2025-07-31:
Apologies if this isn't the ideal place to ask this question.  I've built 2 non-DAC boards, both of which don't seem to do anything at all.  I used to solder professionally and I've tripled check my soldering.  In fact, I built the second board just to see if the behavior was the same and they behave identically.

This is my first pico project but I believe the firmware uploaded properly - I used picotool and it reported back skpico firmware installed.  The pico shows some signs of life - there's a clock signal, I can see what I presume is bus activity from the C64.  SYS 54301 and SYS 54333 just clear the screen and give a READY prompt.  Occasionally they seem to lock up the machine.  SKpicotiming.prg loads and asks for values, then says that it's initializing something (sorry, have to go check that) and just sits there, possibly locked up.

Without a schematic, it's especially hard to know what to look for and where to look for it.  Does anyone have any suggestions for a methodology I can use to troubleshoot these devices?

hpingel, 2025-07-31:
Did you use the green Pico's by Raspberry Pi or the black Pico clones? Sometimes the black clones have soldering imperfections on their PCBs.

pentode1, 2025-07-31:
I used presumably legit picos from Digikey.

Also, in my effort to be concise, I forgot to mention that I see the same behavior on 2 different C64s as well.

LouDnl, 2025-07-31:
Can you post photo's of the bottom of the skpico pcb?

pentode1, 2025-07-31:
This is the completed "bottom." If a couole components look slightly askew, I reflowed them last night and they might have shifted a bit but I've checked for connectivity and shorts and everything seems to check out.

[image]

pentode1, 2025-07-31:
Yes, it's a 2040, and I've double and triple checked the connections on the chips. Definitely no shorts (if you see something that looks like a solder blob it's undoubtedly flux) and I've even traced out every pin I could find on the board and the connectivity is all good.  I've looked at it under stong magnification and all solder joints look good.  

I understand that the skpicotiming would run anyway - I was just mentioning that part to say that it doesn't seem to be communicating with the pico at all.  No sound, no menu.

As I said, I built a second board this morning on the off chance there was a freak issue with one of my solder joints - I think it's unlikely that I ended up with abad connection on both boards.

What I was hoping for was a procedure just to trace out what's alive and what's not, and/ or in the unlikely event I have bad solder joints on both boards,  to trace where the bad connection is.

pentode1, 2025-08-01:
I want to thank everyone for the feedback - I think we can close this topic as I have one board working now.  Funny enough, all the fine pitch stuff was ok, but in case anyone else runs across this, the main problem I had was really poor adhesion on a couple of the SID socket pins.  I'm not sure if it was contamination on the pins I used or something weird about the finish on the boards, but for the record, the pads for those through holes are quite small, so either I didn't get heat in the right spot or... Who knows.  The joints looked ok but the solder just didn't flow well.  I was so convinced something was up with the small stuff I didn't look carefully enough at the easy stuff. 🙂
Oh, and at least one dodgy SID socket.  It's fine with a real SID but for whatever reason doesn't like the round pins.
As usual, multiple problems make for frustrating troubleshooting.

closed 2025-08-01 by pentode1