frntc/SIDKick-pico #122: Multiple compile failures, unable to build

VanessaE, 2025-08-02:
I got my new SIDKickPico 2040DAC in the mail today, and have been trying to get it working.  What I received is almost completely non-functional.  No audio, no config menu<sup>[1]</sup>, no LED activity, no nothing.  I have already reached out to the vendor regarding this, but in the meantime I figured I should at least tinker, in case I can find a solution without waiting for their reply.

I figured, okay maybe this is the timing problem discussed in #113 and elsewhere, so I tried the C64 tool to adjust them, but that tool just hangs after I input the values I want to try.  Or at least, I let it sit for a good solid couple of minutes before giving up.

Okay then, maybe the unit just has no firmware at all, so I soldered-in a suitable connector and made a USB breakout cable according to the details in the README here.  Shorted the "boot" pads, hooked it up, and sure enough the PC could see it, and it popped-up a virtual filesystem.  I copied-over the `SKpico2040DAC.uf2` firmware file from the v0.211 release, and the SKPico seemed to accept it without any problems (if what found by web search is correct<sup>[2]</sup>).  That had no impact though.

Okay, fine, I'll try compiling the firmware from scratch and just put my timings there (well, wherever it is they go -- I didn't get that far), but nope, that did not go well.

You didn't provide any instructions<sup>[3]</sup>, but I figured it out...sorta.  After some web searching, it seemed like I guess these commands should have gotten things going:

```bash
git clone https://github.com/frntc/SIDKick-pico.git SIDKick-pico-source
cd SIDKick-pico-source/Source
cmake -S . -DPICO_EXTRAS_FETCH_FROM_GIT=1 -B build
cmake --build build
```

But nope -- multiple errors shut down that idea:

First there was the error about `audio_spdif.h` not being found, but you told someone else to comment that out, in another post.

Did that, then the next build attempt stopped with `fatal error: hardware/adc.h: no such file` (or words to that effect).  

More web searching led me to `CMakeLists.txt`, seems you/I need to add `hardware_adc` to the `target_link_libraries()` list on line 44.

Fixed that, then on the next attempt died with this...

 `error: 'iobank0_hw' undeclared (first use in this function); did you mean 'io_bank0_hw_t'?`

...and this...

`error: 'padsbank0_hw' undeclared (first use in this function); did you mean 'pads_bank0_hw_t'?`

And that's where I got stuck -- I don't know where to go from here.

-----
[1] I tried both `SYS 54301` and `SYS 54333` and neither works even partially.  In C64 mode, both calls just crash as though the CPU's hitting a BRK instruction.  In C128 mode, both calls break to the built-in ML monitor.

[2] I have never coded anything for the Raspberry Pi, and this is only my second time ever using anything based on it, so I have to go on the assumption that the correct behavior is that the RPi should reset itself after copying a firmware file onto the "pseudo" drive the RPi presents to the system at startup.

[3]  If I'm being honest, the one line comment in the README mentioning that the firmware uses the RPi SDK as part of building it is not very helpful.  You should expand that section, at least spell it out just enough that a RPi nøøb such as me can get through it without heading off to Google.

frntc, 2025-08-02:
The remaining error messages are probably due to changes in the SDK. In any case, the precompiled firmware (which includes the configuration tool you need) normally does work even if the timings are a bit off. You 1) should hear sounds when playing a SID (the config tool might not start reliably), and 2) you can still patch the firmware before uploading with skpicopatch.

frntc, 2025-08-02:
you can look at $D41D-$D41F, you should read { 0x4c, 0x00, 0xd4 }

VanessaE, 2025-08-02:
What it's returning seems... odd:

[image]

frntc, 2025-08-02:
It looks like the SKpico is not communicating properly (in particular if the values change when trying multiple times, i.e. open bus). Have you tried playing a SID? Is there sound?

VanessaE, 2025-08-02:
> It looks like the SKpico is not communicating properly (in particular if the values change when trying multiple times, i.e. open bus). 

The data coming back from `M FD41D` is random/values change, so yeah, "open bus" as you put it.

> Have you tried playing a SID? Is there sound?

Nope -- as I wrote in the opening post, "no audio, no config menu[1], no LED activity, no nothing".

I tried it in my stock, NTSC "flat" C128, but it doesn't work there either.  Same behavior as in the DCR.

VanessaE, 2025-08-03:
At this point I'm declaring my SKPico defective.  I'll try to work something out with the vendor.

In the meantime, I just had a thought regarding the build failures:

> The remaining error messages are probably due to changes in the SDK.

If that is the case, there may be a solution:  you already fetch the SDK and "extras" from git anyway when building, so why not just add the necessary commands to `git checkout` known good commits of each repo, after they're cloned?

That way the user will always be building from known, stable versions.

VanessaE, 2025-08-03:
This should be my final "off-topic" post for this issue.

I got a hunch and tried starting over from scratch, and after doing more power cycles in one day than I normally do in a year 😛 I was able to get my SKPico up to a point where it can show the menu mostly reliably and play music mostly right.

I think the key was the addressing and audio wiring, having the addressing wires connected is not compatible with the SKPico's default setting (second SID at $d400).  

You do mention leaving the wiring disconnected initially, until the second SID setting can be changed in the config menu, but evidently this is far more critical than the README implies:

> Hint: before attaching/detaching signal cables, set the 2nd-SID/FM address to $d400 in the config tool.

I would recommend being more "forceful" with this statement, and I also recommend changing the default second SID to just be disabled.  It already defaults to putting the second SID at $d400, and this appears be INcompatible with having the address wire on an external CS line (in my case, C128 U3 pin 12, which is an unused I/O page at $d700).

So yeah, my problems DO seem to be timing-related.  I can keep tinkering with that, but the values I'm using at this moment are rather low compared to others' numbers.  I think I got it going initially with a readbus value of 6, and a phi value of 10, but I've already forgotten (sorry).

It works, but things are still not 100% right -- for instance, my modplayer thinks the right channel has an 8580 when the SKPico is actually set for 6581on both channels, and it fails to correctly initialize one channel or the other about every 4th time.

So I guess I'll...just shut up about timings/configuring now, that's what #113 is for. 😉

VanessaE, 2025-08-20:
Since you saw fit to lock that other issue, I'll just leave this here:

I wash my hands of this project.   Thanks for your time and good luck in the future.

closed 2025-08-20 by VanessaE